AllMightySauron/homebride-solaxcloud-api #1: Showing Energy and Usage in Homebridge?

jamie-2bit, 2022-01-06:
Does this show usage statistics nativly in Homebridge? All I can see if on/off. Can it use LUX level for example to display Kwh from the inverter?

troygNZ, 2022-01-06:
To evaluate if you like this as a feature, and you think it’s worthwhile adding this - this is what it looks like (including me showing individual string feeds, and non weighted average raw values.

[image]

Screenshot serving from my non cloud version of a Solax Homebridge plugin - https://github.com/troygNZ/homebridge-solax

AllMightySauron, 2022-01-06:
> Does this show usage statistics nativly in Homebridge? All I can see if on/off. Can it use LUX level for example to display Kwh from the inverter?

Hi @jamie-2bit , 

Yes, the plugin provides usage statistics. Take a look at my latest data below :)

[image]

I'm guessing you're using the standard Apple Home app if you can only see on/off characteristics for outlets. Please note that you need to install the [Eve for Homekit] (https://apps.apple.com/us/app/eve-for-homekit/id917695792) or [Controller for HomeKit] (https://apps.apple.com/us/app/controller-for-homekit/id1198176727) app from the App store.

The reason why i'm using these non-standard characteristics is due to the fact that I want to maintain history and see the graphs provided by Eve out-of-the-box.

Regards

AllMightySauron, 2022-01-06:
> To evaluate if you like this as a feature, and you think it’s worthwhile adding this - this is what it looks like (including me showing individual string feeds, and non weighted average raw values. [...]

Hi @troygNZ , I took a look at your plugin before creating this new one.  :wink:

Adding new light accessories with Lux characteristics could overcome the need for using non-standard characteristics for accessories (we could even have both, for that matter :smile:). However, you would lose the ability to oversee charts and history that is offered by the standard Eve history functionality.

This is what I get in Eve, which is basically the same information from your screenshot:

[image]

Hope it helps clarifying the issue.

jamie-2bit, 2022-01-06:
Thanks @AllMightySauron and @troygNZ 

Yes, I'm using the native Apple Home app. I don't personally have the need to see historical data as I can get this in the Solax app itself. I just need real time (5min) data in order to run automations.

Sounds like you too need to combine forces and have the best of both your plugins. Have Solax values showing as LUX levels in the native Apple Home app and have it pulling the values via the Solax API. 

( @troygNZ turns out I can't use your plugin as my inverter has newer firmware and doesn't support the direct API call)

AllMightySauron, 2022-01-06:
> Thanks @AllMightySauron and @troygNZ
> 
> Yes, I'm using the native Apple Home app. I don't personally have the need to see historical data as I can get this in the Solax app itself. I just need real time (5min) data in order to run automations.
> 
> Sounds like you two need to combine forces and have the best of both your plugins. Have Solax values showing as LUX levels in the native Apple Home app and have it pulling the values via the Solax Cloud API.
> 
> ( @troygNZ turns out I can't use your plugin as my inverter has newer firmware and doesn't support the direct API call)

@jamie-2bit , I reckon I can do both. 

What if I add a parameter like, let's say 'Home App Only', and if you set it to true it only creates the Lights with Lux? Would that do the trick?

jamie-2bit, 2022-01-06:
> @jamie-2bit , I reckon I can do both.
>
> What if I add a parameter like, let's say 'Home App Only', and if you set it to true it only creates the Lights with Lux? Would that do the trick?

I think that would be an amazing feature and would help a lot of people with the same use-case as myself! Is it tough to implement?

troygNZ, 2022-01-07:
Feel free to steal my code that surfaces the data as light sensors. It’ll be helpful for those with the cloud version of solax. You may find it useful to provide moving averages on alerts etc, so take fluctuations out when cloud temporarily goes over etc. I have those implemented too. Alternatively, if I can find time I can fork and do a PR back. Disclaimer, I’m not a seasoned NodeJS + TS dev.

AllMightySauron, 2022-01-17:
@troygNZ , @jamie-2bit 

Just to let you know I already have a new version up and running with the option to use ambient light sensors for power. I'll be testing it out over the next couple of days and then publish a beta version for you to test.

Cheers

jamie-2bit, 2022-01-19:
> @troygNZ , @jamie-2bit
> 
> Just to let you know I already have a new version up and running with the option to use ambient light sensors for power. I'll be testing it out over the next couple of days and then publish a beta version for you to test.
> 
> Cheers

How's it been working so far @AllMightySauron ?

AllMightySauron, 2022-01-19:
[image]

Looking good 😄

AllMightySauron, 2022-01-20:
@jamie-2bit , I have just made the beta version available. You can install it by using `npm -g install homebridge-solaxcloud-api@beta`. 

Let me know how it goes...

jamie-2bit, 2022-01-20:
> Let me know how it goes...

Will do, I'll need to wait until the morning for the inverter to kick back in so I'll let you know then. Cheers!

AllMightySauron, 2022-01-21:
> > Let me know how it goes...
> 
> Will do, I'll need to wait until the morning for the inverter to kick back in so I'll let you know then. Cheers!

just to let you know, I have added exponential moving average support on 2.0.2-beta.0

AllMightySauron, 2022-01-24:
> > > Let me know how it goes...
> > 
> > 
> > Will do, I'll need to wait until the morning for the inverter to kick back in so I'll let you know then. Cheers!
> 
> just to let you know, I have added exponential moving average support on 2.0.2-beta.0

@jamie-2bit, any news?

jamie-2bit, 2022-01-24:
> > > > Let me know how it goes...
> > > 
> > > 
> > > Will do, I'll need to wait until the morning for the inverter to kick back in so I'll let you know then. Cheers!
> > 
> > 
> > just to let you know, I have added exponential moving average support on 2.0.2-beta.0
> 
> @jamie-2bit, any news?

Tested it today. It’s working perfectly so far on 2.0.2. Good job!!

AllMightySauron, 2022-01-24:
> Tested it today. It’s working perfectly so far on 2.0.2. Good job!!

Great! I'll move it into production :smile:

closed 2022-01-24 by AllMightySauron

troygNZ, 2022-01-24:
Great work. I only have the non cloud variant, but great stuff!