AllMightySauron/homebride-solaxcloud-api #2: Please add second inverter and Battery charge percent

JanHios, 2022-03-28:
Many Solax users have more than one inverter. Can you please add a second inverter ( api data of both inverter needs to be summarized)
And there are lots of users that have a battery installed at their first inverter. 
the API output lists batt charge percentage at the label “soc” and the charge or discharge amount at the label “batPower”
Tgank you for the great work

AllMightySauron, 2022-03-29:
Sure, I can add the option to support multiple inverters. 

As far as batteries, I do not have one installed right now. I'll need help with some examples of the API results for computing battery math. Are you able to assist?

JanHios, 2022-03-30:
This is the API output with batt installed: ( SOC = batt charge percentage / batPower = charge or discharge Watts of the battery )

{"success":true,"exception":"Query success!","result":{"inverterSN":"Xxxxxxxxxxxxx","sn":"Xxxxxxxxxxxx","acpower":-369.0,"yieldtoday":4.4,"yieldtotal":6628.9,"feedinpower":-63.0,"feedinenergy":4777.96,"consumeenergy":4688.68,"feedinpowerM2":0.0,"soc":53.0,"peps1":0.0,"peps2":0.0,"peps3":0.0,"inverterType":"5","inverterStatus":"102","uploadTime":"2022-03-30 10:25:19","batPower":1949.0,"powerdc1":1061.0,"powerdc2":572.0,"powerdc3":null,"powerdc4":null}}

AllMightySauron, 2022-03-30:
Thanks, I'll let you know as soon as I have a new version released for you to test...

AllMightySauron, 2022-03-30:
You can try out the new beta with support for multiple inverters with `npm install homebridge-solaxcloud-api@2.1.1-beta.0`. It will be easier to setup using homebridge-ui...

AllMightySauron, 2022-03-31:
> Hi. Yes it makes sense to show the total sum of all inverter numbers ( same does Solax in their App ). I would show the sum of solar out , total of grid feed in / consumption , the sum of Batt charge / discharge and finally the sum of power usage of your home. Also the batt percentage shown as a sensor would be very helpful for home automation. I did the same at a iOS App for the Solax api : https://www.solxview.de/en You can test it for free.

Sure, I'll start adding the new accessories and create a "virtual" inverter which sums figures for all the others.

Did you have the chance to test out the new version for both your inverters? Did it work ok?

JanHios, 2022-03-31:
Yes it worked fine. 
[image]

AllMightySauron, 2022-03-31:
> SOC

Great, moved out of the beta tag. Working on the remaining sensors for batteries...

AllMightySauron, 2022-04-04:
@JanHios , can you please try out the latest beta?

It includes battery support and a "virtual inverter" for the totals. Battery SoC is still missing, though...

JanHios, 2022-04-04:
It doesn’t show summarised data:
[image]
[image]
[image]

JanHios, 2022-04-04:
2.1.4 beta is correct ?

JanHios, 2022-04-04:
Bat is now switched on and shows up. Today is bad weather here in Germany. Our Batt is empty. I have to wait until we have more sunshine again. I will check it asap.

AllMightySauron, 2022-04-04:
ok, can you see a line containing "Computing inverter totals..." in your homebridge log?

JanHios, 2022-04-05:
Yes I found it. But there is another issue when initialising the totals:

[image]
[image]

JanHios, 2022-04-05:
What I also see, is that the individual inverter numbers do not match the sum that the Solax app shows. 
find here my api outputs and a Solax app screenshot and the your output. 

[image]
[image]
[image]
[image]
[image]

JanHios, 2022-04-05:
It looks like the powerDC output of the solar panels are not correctly summarised. Also the other numbers are different.

AllMightySauron, 2022-04-05:
My bad, sumarized values for the virtual inverter "All inverters" should be fixed in the latest beta. I have just made it available.

As far as the powerDC output, let me check it...

AllMightySauron, 2022-04-05:
> Yes I found it. But there is another issue when initialising the totals:
> 
> [image] [image]

The error above typically happens when the plugin is unable to contact SolaxCloud during initialization... Don't worry, it should recover automatically afterwards.

JanHios, 2022-04-05:
Yes it did. Now the total PV power , grid power and batt in/out are correct. There is still a difference concerning for the “to house sensor” compared with the Solax app. I try to figure out what it could be…

JanHios, 2022-04-05:
I have checked my notes of my SolXView App:

due to the fact that the solar ACpower values turn negative at inverter that charge batteries, it is not really reliable.
Also there is a special case if Batt is empty.
My approach was the following:

// Doku:
 In general we have to  take for total HouseConsumption instead of "Act Solar Output power" the "Invacpower Minus the feedInPower" values. Difference between Solar out and Inverter Out ( ACPower)  is the energy that is lost at  the inverter itslf ( e.g. heat) ( This way the loss of the Inverters is also taken  into account and indication is same like at the Solax App. There is just one additionally case to cover if the Batt is empty 

RMK: The following is related to already summmarised figures of all inverter:

// for all following calculations make Bat power always positive
            if BatPowerTemp <= 0 {
            BatPowerTemp = BatPowerTemp * -1
            }

// if Bat Power < 11 und Bat Percent is 10 or below (= battery empty 
( RMK: Solax keeps 10 percent as minimum charge to protect battery))  
basically we have no batt output - so use grid power plus solar power
            
            if ((BatPowerTemp <= 10 ) && (self.Invsoc <= 11 ) ){
                // make grid power always positive 
                
                var InvfeedinpowerTemp = self.Invfeedinpower
                if InvfeedinpowerTemp <= 0 {
                    InvfeedinpowerTemp = InvfeedinpowerTemp * -1
                }
                
                // now calculate total consumption =  grid power plus total solar output
                
                var HouseConsumptionTemp = InvfeedinpowerTemp + self.ActSolarOutput
                
                if HouseConsumptionTemp  <= 0 {
                    HouseConsumptionTemp  = HouseConsumptionTemp  * -1
                }
                

// else (RMK batt is not empty)
            
            // make grid power always positive 
                
                var InvfeedinpowerTemp = self.Invfeedinpower
                if InvfeedinpowerTemp <= 0 {
                    InvfeedinpowerTemp = InvfeedinpowerTemp * -1
                }
                
                var HouseConsumptionTemp = self.Invacpower - self.Invfeedinpower
                if HouseConsumptionTemp <= 0 {
                    HouseConsumptionTemp  = HouseConsumptionTemp  * -1
                }
                

                 
I hope this helps you. It took a while until I figured this out, so hopefully it saves you some time ;-)

AllMightySauron, 2022-04-07:
@JanHios , you can install the latest official release. It already includes battery SoC information.

Now, if my calculations are correct, I just need to recheck AC Power... :smile:

JanHios, 2022-04-07:
Works great. The only thing that I found was that the Batt % module that you have setup 
[image]

can not be used for automation. Because the batt status of a device is not usable as trigger. 

I would suggest to use a humidity module. That provides percentage from 0 to 100 and can be used as a trigger. 
[image]

thanks a lot for your great work.

AllMightySauron, 2022-04-07:
Nothing is ever easy, right? :smile:

BTW, I have added battery SoC totals to the virtual "All inverters" inverter just in case you have more than on battery. Charge state is set to "Charging" if at least one battery is charging and I'm averaging the individual battery levels to summarize those onto a single value. In your case, it should be equal to the individual inverter battery figures...

AllMightySauron, 2022-04-07:
Just to let you know: you can create an automation based on the battery level with the current accessory by using Eve...

Use the motion detector as a trigger:
[image]

and set a condition using the battery level:
[image]

closed 2022-05-03 by AllMightySauron

JanHios, 2022-05-15:
> @JanHios<https://github.com/JanHios> , opened a new issue for the "special cases" above. I'm gonna need you help providing data when the error occurs, since I cannort seem to reproduce it...

Hi,

I am on vacation and will be back next week.
I will help you as soon as I am up to date on my IT job.

Cheers Jan

JanHios, 2022-07-20:
Sorry for my late replay. I had so
Much to do on my job. 
I am using your plugin now continuously and did not had any issues anymore except the following, that caused the plug-in to stop transferring data since a few weeks. It only helps to restart the Homebridge. 
see picture of the log:
[image]

JanHios, 2022-07-20:
Sorry here is the picture:
[image]