AllMightySauron/homebride-solaxcloud-api #4: Plugin fails after some time

JanHios, 2022-07-20:
**Describe The Bug:**
the plug-in stops working from time to time with the following error in the log. Only a Homebridge restart helps. Other plug-ins continue to work in parallel. Only the Solax plug-in stops working
[image]

**To Reproduce:**
difficult to say - it suddenly started. 

**Expected behavior:**
work normally 

**Logs:**
See screenshot above

AllMightySauron, 2022-07-20:
@JanHios , can you please test out version [2.1.8](https://github.com/AllMightySauron/homebride-solaxcloud-api/releases/tag/2.1.8)?

I have simplified the data pooling sleep mechanism. I hope this solves it...

JanHios, 2022-07-22:
I have installed it right after your message and until now it works fine. Let me test it for some more days, but of you don’t hear anything from me until end of next week it is working fine.

closed 2022-07-29 by AllMightySauron