AllMightySauron/homebride-solaxcloud-api #5: Please include Q-Cells Cloud (branded version of SolaX)

rudibarani, 2023-07-18:
**Is your feature request related to a problem? Please describe:**
Q-Cells is a global retailer of SolaX-Products which uses the same infrastructure and software except these are branded differently. Please include their "products" into your Homebridge plugin, too.

**Describe the solution you'd like:**
As the backend & software is the same, it should boil down to using the Q-Cells API and server to include the Q-Cells branded line into homebridge. The URL-structure is like this:
https://www.portal-q-cells.us/proxyApp/proxy/api/getRealtimeInfo.do?tokenId=XXX&sn=YYY
Maybe an option to select between the two different servers would be possible?

I am happy to be a test user to get his running :)

AllMightySauron, 2023-07-20:
Hi @rudibarani , glad to help. If it's only a matter of changing the URL, it shouldn't be that much effort.

I have no way of testing it though...

rudibarani, 2023-07-26:
Thanks for the reply and interest in integrating Q-Cells into your Plugin.

I tried the changed URL in a fork and it works 1:1 as described by you for Solax. The accessories behave like they should and show all the details you mentioned in the Eve-App.

There was only one issue with naming of the smooth sensors. The Apple Home App told me that their names cannot end with a ")" - only letters and numbers seem to be allows as first or last symbol in the current version of HomeKit. 

It would be great if you could add an option to your Plugin to define if the inverter is branded by Solax or Q-Cells and use the respective URL.

AllMightySauron, 2023-07-29:
Hi,

Glad to hear that.

Can you try out the new beta by installing the plugin with `npm install homebridge-solaxcloud-api@2.2.1-beta.0 `?

Thanks

rudibarani, 2023-07-29:
Thank you very much. I'm on vacation for a week and look forward to trying it the week after.

AllMightySauron, 2023-08-14:
Hi, any news regarding testing of this version?

rudibarani, 2023-08-14:
Hi.
We are back home and I got to try it today. It works perfectly fine with my Q-Cells inverter. Thank you very much for integrating Q-Cells into your plugin!

There is one remaining issue when installing the "smooth" devices. The Home-App on my iPhone told me for each device that "... (smooth)" could no be saved. Their names must start and end with a letter or number and cannot end with a special character.
Another suggestion would be to change the default name for the homebridge-devices. If one does not change it from "homebride-solaxcloud-api" to e.g. Solax or Q-Cells, each new homebridge-device starts with this long name.

Best,
Phillip

closed 2024-03-12 by AllMightySauron